noble/bleno #416: Advertising not starting in macos Mojave

adotg, 2018-10-06:
Using bleno as sandalone module to mock a peripheral

Steps of running: 
- clone the repo
- Navigate inside bleno/examples/echo
- node main.js

Console shows `on -> stateChange: poweredOn` but `on -> advertisingStart: success` not coming.

Debug log
```
bleno - echo
  bleno platform darwin +0ms
  bleno addressChange 8c:85:90:64:f0:b4 +3ms
  highsierra-bindings sendCBMsg: 1, {
  highsierra-bindings   "kCBMsgArgName": "node-1538851048987",
  highsierra-bindings   "kCBMsgArgOptions": {
  highsierra-bindings     "kCBInitOptionShowPowerAlert": 1
  highsierra-bindings   },
  highsierra-bindings   "kCBMsgArgType": 1
  highsierra-bindings } +0ms
  highsierra-bindings xpcEvent: {
  highsierra-bindings   "kCBMsgId": 4,
  highsierra-bindings   "kCBMsgArgs": {
  highsierra-bindings     "kCBMsgArgState": 5
  highsierra-bindings   }
  highsierra-bindings } +1ms
  highsierra-bindings state change poweredOn +0ms
  bleno stateChange poweredOn +2ms
on -> stateChange: poweredOn
startAdv { kCBAdvDataLocalName: 'echo',
  kCBAdvDataServiceUUIDs: [ <Buffer ec 00> ] }
  highsierra-bindings sendCBMsg: 16, {
  highsierra-bindings   "kCBAdvDataLocalName": "echo",
  highsierra-bindings   "kCBAdvDataServiceUUIDs": [
  highsierra-bindings     {
  highsierra-bindings       "type": "Buffer",
  highsierra-bindings       "data": [
  highsierra-bindings         236,
  highsierra-bindings         0
  highsierra-bindings       ]
  highsierra-bindings     }
  highsierra-bindings   ]
  highsierra-bindings } +3ms
```

Environment MacOS Mojave version 10.14

rafaelbuiu, 2019-03-13:
Did you guys find a solution for that? I am not able to find make it.. I have tried everything on the web. Not even an other library for creating peripherals on MacOSX

LukasBombach, 2019-07-04:
Hey guys, I made a working version, see my repo at this commit:

https://github.com/LukasBombach/de1-emulator/tree/6d31cd4fa51dbe3c0f0bf039f0bc20bebf85c326

basically you need this `package.json`:

```js
{
  "dependencies": {
    "bleno": "github:notjosh/bleno#inject-bindings",
    "bleno-mac": "github:notjosh/bleno-mac",
    "xpc-connection": "sandeepmistry/node-xpc-connection#pull/26/head"
  },
  "resolutions": {
    "xpc-connection": "sandeepmistry/node-xpc-connection#pull/26/head"
  }
}
```

and then you can run the echo example: https://github.com/noble/bleno/tree/master/examples/echo

See my whole commit here: https://github.com/LukasBombach/de1-emulator/commit/6d31cd4fa51dbe3c0f0bf039f0bc20bebf85c326

charliegerard, 2019-07-11:
Thanks @LukasBombach for your version!! It works for me, but only if I use the default version of Python installed on my Mac...
For some reason that I don't understand yet, if I'm using Python 3, I still run into errors when running `npm install` :/

LukasBombach, 2019-08-07:
I got stuck, so I started completely reimplementing Noble. You can follow my progress here:

https://github.com/LukasBombach/sblendid/

Current status: everything seems to work fine but I am developing on macOS (Mojave) only right now and I need to add support for other platforms

farfromrefug, 2019-10-16:
To anyone interested none of the solutions were working for me anymore (mostly due to versions update).
I managed to update bleno-mac to make it work though.
```
"dependencies": {
        "bleno-mac": "github:farfromrefug/bleno-mac"
    }
```

farfromrefug, 2019-10-16:
@LukasBombach BTW would love to help you make a new bleno plugin like you did with noble. Let me know if i can help

LukasBombach, 2019-10-16:
@farfromrefug sure, thanks. BTW there is

https://github.com/notjosh/bleno-mac

which is a native implementation for bleno on macOS, not sure if that helps you. it worked for me though.

farfromrefug, 2019-10-16:
@LukasBombach yes that s what i am using (modified version though). But as it is not maintained and as i like your adpater approach i though a new plugin would be lovely :D

LukasBombach, 2019-10-16:
Of couse. For now I want to finish Sblendid first, to improve what Noble is lacking, I don't think I will put my hands on Bleno any time soon though.