noble/bleno #421: Broadcasting data 60 FPS with GATT

oishi89, 2018-12-18:
Hello.
I'm using a CSR V.4 (csr8510 a10) and I'm trying to broadcast 40 bytes of data with 40 - 60 time per second. It seems to be possible and working fine but after sending data for a while I got a disconnection randomly with LIB USB ERROR message. I don't know how to figure out the throughput or bandwidth of CSR V.4. Do you have any ideal about that?
Thanks

zach-luman, 2018-12-19:
Hi @oishi89,
How do you broadcast the data?
Could you please upload the relevant code snippets?

oishi89, 2018-12-19:
@zach-luman I was creating a timer with interval time is 16 ms.

zach-luman, 2018-12-19:
In onSubscribe handler? @oishi89

zach-luman, 2018-12-19:
Please upload the full characteristic code here.

oishi89, 2018-12-19:
`const bleno = require("bleno");

console.log("Starting bleno...");

const COUNTER_SERVICE_UUID = "00010000-9FAB-43C8-9231-40F6E305F96D";
const COUNTER_CHAR_UUID = "00010001-9FAB-43C8-9231-40F6E305F96D";

class CounterCharacteristic extends bleno.Characteristic {
    constructor() {
        super({
            uuid: COUNTER_CHAR_UUID,
            properties: ["notify"],
            value: null
        });

        this.counter = 0;
    }

    onSubscribe(maxValueSize, updateValueCallback) {
        console.log(`Counter subscribed, max value size is ${maxValueSize}`);
        this.updateValueCallback = updateValueCallback;
    }

    onUnsubscribe() {
        console.log("Counter unsubscribed");
        this.updateValueCallback = null;
    }    

    sendNotification(value) {
        if(this.updateValueCallback) {
            console.log(`Sending notification with value ${value}`);

            const notificationBytes = new Buffer(4 * 10);

            for (var i = 0; i < 10; i++) {
                    notificationBytes.writeFloatLE(value * 0.01, i * 4);        
            }

            this.updateValueCallback(notificationBytes);
        }
    }

    start() {
        console.log("Starting counter");
        this.handle = setInterval(() => {
            this.counter = (this.counter + 1) % 0xFFFF;
            this.sendNotification(this.counter);
        }, 16);
    }

    stop() {
        console.log("Stopping counter");
        clearInterval(this.handle);
        this.handle = null;
    }
}

let counter = new CounterCharacteristic();
counter.start();

bleno.on("stateChange", state => {

    if (state === "poweredOn") {
        
        bleno.startAdvertising("Counter", [COUNTER_SERVICE_UUID], err => {
            if (err) console.log(err);
        });

    } else {
        console.log("Stopping...");
        counter.stop();
        bleno.stopAdvertising();
    }        
});

bleno.on("advertisingStart", err => {

    console.log("Configuring services...");
    
    if(err) {
        console.error(err);
        return;
    }

    let service = new bleno.PrimaryService({
        uuid: COUNTER_CHAR_UUID,
        characteristics: [counter]
    });

    bleno.setServices([service], err => {
        if(err)
            console.log(err);
        else
            console.log("Services configured");
    });
});

// some diagnostics 
bleno.on("stateChange", state => console.log(`Bleno: Adapter changed state to ${state}`));

bleno.on("advertisingStart", err => console.log("Bleno: advertisingStart"));
bleno.on("advertisingStartError", err => console.log("Bleno: advertisingStartError"));
bleno.on("advertisingStop", err => console.log("Bleno: advertisingStop"));

bleno.on("servicesSet", err => console.log("Bleno: servicesSet"));
bleno.on("servicesSetError", err => console.log("Bleno: servicesSetError"));

bleno.on("accept", clientAddress => console.log(`Bleno: accept ${clientAddress}`));
bleno.on("disconnect", clientAddress => console.log(`Bleno: disconnect ${clientAddress}`));`

zach-luman, 2018-12-20:
@oishi89 
Thanks for uploading code snippet.
But I'm wondering how did you let this peripheral subscribed?
In other word, how is onSubscribe handler called?