noble/bleno #425: Cannot install bleno with nodejs 10.x in a raspberry pi zero. V8_DEPRECATED error

nikolaof, 2018-12-31:
Hi,

I'm trying to install the bleno module for node 10.15.0 using either a `package.json` file or just by running the `npm install bleno` I'm getting the following error:

```
 binding.target.mk:99: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/dev/nodejs/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/opt/nodejs/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
gyp ERR! System Linux 4.14.79+
gyp ERR! command "/opt/nodejs/bin/node" "/opt/nodejs/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/dev/nodejs/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN ble-server@1.0.0 No description
npm WARN ble-server@1.0.0 No repository field.
npm WARN ble-server@1.0.0 license should be a valid SPDX license expression
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
```

It seems that it's trying to use an old bluetooth-hci-socket version that using the deprercated `V8` macro

While compiling it emmited this error:

```
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
```

Any idea? 
Maybe the package.json should be changed in order to use the `@abandonware/bluetooth-hci-socket` module.

xiongsongsong, 2019-01-22:
Yes, I also met.

Webunity, 2019-02-20:
This is not an issue with bleno but with one of the used packages:
https://github.com/noble/node-bluetooth-hci-socket/issues/84
https://github.com/noble/node-bluetooth-hci-socket/pull/91

schwamic, 2019-03-22:
An update to Node10 would be great!

arthare, 2020-08-17:
A quick solution is to point bleno's package.json dep for node-bluetooth-hci-socket to `"@abandonware/bluetooth-hci-socket": "^0.5.3-5"`

Got it building on my raspi 4, after having the exact error posted in the OP.

https://github.com/arthare/bleno/commit/596acd9f32f889b8757f66bde23e25f488644259

nkemcels, 2020-11-12:
I'm using `@abandonware/bleno` with node.js `12.19` and so far everything works well.

Dizi0, 2021-01-07:
I followed @arthare advices, I'm running it with NodeJs 15 like a charm, thank you.