noble/bleno #439: npm install bleno fail

ervinewell, 2019-08-01:
npm i xpc-connection

> xpc-connection@0.1.4 install /Users/ervinewell/Desktop/node-test/node_modules/xpc-connection
> node-gyp rebuild

gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
  CXX(target) Release/obj.target/binding/src/XpcConnection.o
../src/XpcConnection.cpp:103:41: warning: 'IntegerValue' is deprecated
      [-Wdeprecated-declarations]
    xpcObject = xpc_int64_create(value->IntegerValue());
                                        ^
/Users/ervinewell/.node-gyp/10.16.0/include/node/v8.h:2476:3: note:
      'IntegerValue' has been explicitly marked deprecated here
  V8_DEPRECATED("Use maybe version", int64_t IntegerValue() const);
  ^
/Users/ervinewell/.node-gyp/10.16.0/include/node/v8config.h:327:29: note:
      expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/XpcConnection.cpp:113:40: warning: 'ToObject' is deprecated
      [-Wdeprecated-declarations]
    Local<Object> valueObject = value->ToObject();
                                       ^
/Users/ervinewell/.node-gyp/10.16.0/include/node/v8.h:2455:10: note: 'ToObject'
      has been explicitly marked deprecated here
  inline V8_DEPRECATED("Use maybe version", Local<Object> ToObject() const);
         ^
/Users/ervinewell/.node-gyp/10.16.0/include/node/v8config.h:327:29: note:
      expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/XpcConnection.cpp:123:40: warning: 'ToObject' is deprecated
      [-Wdeprecated-declarations]
    Local<Object> valueObject = value->ToObject();
                                       ^
/Users/ervinewell/.node-gyp/10.16.0/include/node/v8.h:2455:10: note: 'ToObject'
      has been explicitly marked deprecated here
  inline V8_DEPRECATED("Use maybe version", Local<Object> ToObject() const);
         ^
/Users/ervinewell/.node-gyp/10.16.0/include/node/v8config.h:327:29: note:
      expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/XpcConnection.cpp:143:89: error: too few arguments to function call,
      expected 2, have 1
  ...propertyValue = object->GetRealNamedProperty(propertyName->ToString());
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~                         ^
/Users/ervinewell/.node-gyp/10.16.0/include/node/v8.h:3562:3: note:
      'GetRealNamedProperty' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<Value> GetRealNamedProperty(
  ^
/Users/ervinewell/.node-gyp/10.16.0/include/node/v8config.h:416:31: note:
      expanded from macro 'V8_WARN_UNUSED_RESULT'
#define V8_WARN_UNUSED_RESULT __attribute__((warn_unused_result))
                              ^
../src/XpcConnection.cpp:254:12: warning: 'MakeCallback' is deprecated
      [-Wdeprecated-declarations]
      Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").T...
           ^
../../nan/nan.h:1045:3: note: 'MakeCallback' has been explicitly marked
      deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
  ^
../../nan/nan.h:104:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/XpcConnection.cpp:263:12: warning: 'MakeCallback' is deprecated
      [-Wdeprecated-declarations]
      Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").T...
           ^
../../nan/nan.h:1045:3: note: 'MakeCallback' has been explicitly marked
      deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
  ^
../../nan/nan.h:104:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
5 warnings and 1 error generated.
make: *** [Release/obj.target/binding/src/XpcConnection.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Darwin 18.0.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/ervinewell/Desktop/node-test/node_modules/xpc-connection
gyp ERR! node -v v10.16.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN test@1.0.0 No description
npm WARN test@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})

os: macOS 10.14
node: v10.16.0

matzuk, 2019-09-18:
I think that the answer is obvious. The library is not supported very well at last time. That's why you need to use a lower version of NodeJs.

MrZMN, 2019-12-06:
I faced the same problem. The problem is that the version of Nodejs is too high. I sassiest you to use a lower Nodejs version (Mine is node -v9.11.2). To do this, try to install nvm and you'll find it extremely easy to use this tool to manage your versions (you can find tutorials on the Internet or simply use command 'nvm help')