noble/bleno #440: Cannot find module 'bluetooth-hci-socket'

ElesGergo, 2019-09-02:
**I am trying to make bleno work with a raspberry pi 3 and I am getting this error:**
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'bluetooth-hci-socket'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/home/pi/Desktop/hwTest/node_modules/bleno/lib/hci-socket/mgmt.js:6:26)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)

**I had the same issue on windows but I manged to slove that by installing the package however here when I am trying to install the missing module I am getting this error:**
 npm i bluetooth-hci-socket
npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

> usb@1.6.0 install /home/pi/Desktop/code/ent_gate/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.0
prebuild-install info looking for cached prebuild @ /home/pi/.npm/_prebuilds/2643fa-usb-v1.6.0-node-v64-linux-arm.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.0/usb-v1.6.0-node-v64-linux-arm.tar.gz
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.0/usb-v1.6.0-node-v64-linux-arm.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=10.15.2 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/home/pi/Desktop/code/ent_gate/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from ../src/helpers.h:3,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../node_modules/nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../node_modules/nan/nan.h:2232:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
In file included from ../src/node_usb.h:14,
                 from ../src/node_usb.cc:1:
../src/node_usb.cc: At global scope:
/usr/include/nodejs/src/node.h:570:43: warning: cast between incompatible function types from ‘void (*)(v8::Local<v8::Object>)’ to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
       (node::addon_register_func) (regfunc),                          \
                                           ^
/usr/include/nodejs/src/node.h:604:3: note: in expansion of macro ‘NODE_MODULE_X’
   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
   ^~~~~~~~~~~~~
../src/node_usb.cc:95:1: note: in expansion of macro ‘NODE_MODULE’
 NODE_MODULE(usb_bindings, Initialize)
 ^~~~~~~~~~~
../src/node_usb.cc: In function ‘void handleHotplug(std::pair<libusb_device*, libusb_hotplug_event>)’:
../src/node_usb.cc:151:58: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
  Nan::MakeCallback(Nan::New(hotplugThis), "emit", 2, argv);
                                                          ^
[... 218 lines cut ...]
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.19.69-v7+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/Desktop/code/ent_gate/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN gate@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the bluetooth-hci-socket@0.5.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-02T16_14_06_552Z-debug.log

**Any suggestions?**

ElesGergo, 2019-09-06:
Resolved by downgrading nodejs

closed 2019-09-06 by ElesGergo

timothyparez, 2020-01-02:
@ElesGergo could you tell us to what version exactly, I am having the same issue.

gnnsekhar, 2020-02-05:
@ElesGergo could you tell us to what version exactly, I am having the same issue.

vrdeepakprakash, 2020-03-12:
Install node version manager:

` sudo npm install -g n`

Downgrade node to v 8.9.0 by:

`sudo n 8.9.0`

then

`sudo npm install bluetooth-hci-socket --unsafe-perm`

nkemcels, 2020-11-12:
Or if downgrading nodejs is not an option for you as it was in my case, then use `@abandonware/bleno` instead.

abdallahdhr, 2024-04-01:
Download the library  @abandonware/noble  into the project folder  and run the file script .