noble/bleno #452: value sent several times on read request

abelhoui, 2020-05-04:
Hi all,

I've been working on bleno to implement a peripheral BLE device, and i'm now facing a problem when i execute the read request from my phone using a ble scanner app, to read a characteristic value from my peripheral device which is a rpi running a BLE peripheral demo using bleno : the value is sent several times when the value size exceeds some particular number of bytes !!!
 - Here i present two cases where one is working fine and the second is beiig sent several times:
++++exemple 1:

`var vehicle_id = 'ivESK_Wagen 1.03'

var vehicle_id_buf = Buffer.from(vehicle_id);

VehicleIDCharacteristic.prototype.onReadRequest = function(offset, callback) {
callback(this.RESULT_SUCCESS, vehicle_id_buf);
}; 
`
++++exemple 2:

`var vehicle_id = 'ivESK_Wagen 1.03 : here we are running a read test for the vehicle_id read operation'

var vehicle_id_buf = Buffer.from(vehicle_id);

VehicleIDCharacteristic.prototype.onReadRequest = function(offset, callback) {
callback(this.RESULT_SUCCESS, vehicle_id_buf);
}; 
`

- Is there a specification for a limited number of bytes to be sent? am I missing sth?

Thanks !

aptarmy, 2020-10-02:
Any update on this? I am facing the same issue on my Raspi Zero W. With few bytes response `onRead`, the library works very well, but if Raspi peripheral response is exceeded some number of bytes, `onRead` method get called several times.

abelhoui, 2020-10-05:
> Any update on this? I am facing the same issue on my Raspi Zero W. With few bytes response `onRead`, the library works very well, but if Raspi peripheral response is exceeded some number of bytes, `onRead` method get called several times.

Heyy @aptarmy yeah that's due to Ble limitation you cannot send more than 21 bytes at time so you should use fragmentation for that. check the link below:
https://github.com/noble/bleno/blob/master/test.js#L36

aptarmy, 2020-10-16:
Thank you so much