noble/bleno #461: Getting error on disconnect

waghekapil, 2021-02-18:
Hello,

I am using this library on RockPi 4B v1.3 and OS is [Debian 9 Desktop](https://wiki.radxa.com/Rockpi4/Debian).

On disconnect I'm getting below error. We have used nRF Connect App, our own mobile apps for the testing. With all the apps we are getting same error. If I moved the device out of the range, I get the same error.

`bleno.on('disconnect', callback (clientAddress)` also not worked.

Can you please check and help me to handle it error or solve the issue.

`/home/linaro/SM/node_modules/bleno/lib/hci-socket/gatt.js:283
        this._handles[i].uuid === '2902' && this._handles[i].value.readUInt16LE(0) !== 0) {
                                                                   ^

TypeError: this._handles[i].value.readUInt16LE is not a function
    at Gatt.onAclStreamEnd (/home/linaro/SM/node_modules/bleno/lib/hci-socket/gatt.js:283:6
    at emitNone (events.js:111:20)
    at AclStream.emit (events.js:208:7)
    at AclStream.push (/home/linaro/SM/node_modules/bleno/lib/hci-socket/acl-stream.js:28:1
    at BlenoBindings.onDisconnComplete (/home/linaro/SM/node_modules/bleno/lib/hci-socket/b
154:21)
    at emitTwo (events.js:126:13)
    at Hci.emit (events.js:214:7)
    at Hci.onSocketData (/home/linaro/SM/node_modules/bleno/lib/hci-socket/hci.js:520:12)
    at emitOne (events.js:116:13)
    at BluetoothHciSocket.emit (events.js:211:7)`

waghekapil, 2021-02-24:
Can you please help me?

Han-HeyGyeong, 2021-08-19:
you still have this problem? I have got same thing. 
so I remove && after code. 

`this._handles[i].value.readUInt16LE(0) !== 0`

YOURDIRECTORY/node_modules/bleno/lib/hci-socket/gatt.js:283

No other problems occurred. But I'm not 100% sure.