noble/bleno #475: disconnect from service after arbitrary amount of time

dewetdt, 2022-01-19:
Hi I am are writing a service on a rpi CM4, the BLE connection drops after a set amount of time. 

btmon log:
```
0.223304
= Note: Bluetooth subsystem version 2.22                                                                                                                                            0.223317
= New Index: E4:5F:01:2C:B2:6D (Primary,UART,hci0)                                                                                                                           [hci0] 0.223326
= Open Index: E4:5F:01:2C:B2:6D                                                                                                                                              [hci0] 0.223328
= Index Info: E4:5F:01:2C:B2:6D (Cypress Semiconductor)                                                                                                                      [hci0] 0.223332
@ RAW Open: node (privileged) version 2.22                                                                                                                          {0x0002} [hci0] 0.223336
@ MGMT Open: node (privileged) version 1.18                                                                                                                                {0x0001} 0.223340
> HCI Event: LE Meta Event (0x3e) plen 19                                                                                                                                #1 [hci0] 46.387030
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 64
        Role: Slave (0x01)
        Peer address type: Random (0x01)
        Peer address: 48:A5:28:E6:17:D1 (Resolvable)
        Connection interval: 45.00 msec (0x0024)
        Connection latency: 0 (0x0000)
        Supervision timeout: 5000 msec (0x01f4)
        Master clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 13                                                                                                                    {0x0001} [hci0] 46.387146
        LE Address: 48:A5:28:E6:17:D1 (Resolvable)
        Flags: 0x00000000
        Data length: 0
> HCI Event: Vendor (0xff) plen 5                                                                                                                                        #2 [hci0] 46.387042
        55 00 00 40 00                                   U..@.
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2                                                                                                         #3 [hci0] 46.387420
        Handle: 64
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                #4 [hci0] 46.388021
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
> HCI Event: Command Complete (0x0e) plen 14                                                                                                                             #5 [hci0] 46.388031
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
        00 00 00 00 00 00 00 00 00 00                    ..........
> HCI Event: LE Meta Event (0x3e) plen 12                                                                                                                                #6 [hci0] 46.456079
      LE Read Remote Used Features (0x04)
        Status: Success (0x00)
        Handle: 64
        Features: 0x3f 0x00 0x00 0x00 0x00 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Slave-initiated Features Exchange
          LE Ping
          LE Data Packet Length Extension
> ACL Data RX: Handle 64 flags 0x02 dlen 11                                                                                                                              #7 [hci0] 46.815953
      ATT: Read By Group Type Request (0x10) len 6
        Handle range: 0x0001-0xffff
        Attribute group type: Primary Service (0x2800)
< ACL Data TX: Handle 64 flags 0x00 dlen 24                                                                                                                              #8 [hci0] 46.816698
      ATT: Read By Group Type Response (0x11) len 19
        Attribute data length: 6
        Attribute group list: 3 entries
        Handle range: 0x0001-0x0005
        UUID: Generic Access Profile (0x1800)
        Handle range: 0x0006-0x0009
        UUID: Generic Attribute Profile (0x1801)
        Handle range: 0x000a-0x0016
        UUID: Device Information (0x180a)
> ACL Data RX: Handle 64 flags 0x02 dlen 11                                                                                                                              #9 [hci0] 46.995356
      ATT: Read By Group Type Request (0x10) len 6
        Handle range: 0x0017-0xffff
        Attribute group type: Primary Service (0x2800)
< ACL Data TX: Handle 64 flags 0x00 dlen 26                                                                                                                             #10 [hci0] 46.996037
      ATT: Read By Group Type Response (0x11) len 21
        Attribute data length: 20
        Attribute group list: 1 entry
[... 467 lines cut ...]
      LE Connection Update Complete (0x03)
        Status: Success (0x00)
        Handle: 64
        Connection interval: 45.00 msec (0x0024)
        Connection latency: 0 (0x0000)
        Supervision timeout: 5000 msec (0x01f4)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                                         #109 [hci0] 50.516358
        Handle: 64
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                              #110 [hci0] 50.516711
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                         #111 [hci0] 50.641241
        Status: Success (0x00)
        Handle: 64
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                  {0x0001} [hci0] 50.641289
        LE Address: 48:A5:28:E6:17:D1 (Resolvable)
        Reason: Connection terminated by local host (0x02)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                                                                                                            #112 [hci0] 50.672496
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                            #113 [hci0] 50.672930
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
```

josiaheakle, 2022-06-09:
I am dealing with the same issue. The connection stays open for maybe 30 seconds, but will end even if a service is being used.

jayjoshi1995, 2022-06-27:
so how can resolve this issue.

rzr, 2023-06-30:
May I suggest to try @abandonware fork and report at:
Relate-to: https://github.com/abandonware/bleno/issues/17